spacetelescope/stcal #42: Ramp fit with multiprocessing hangs on macOS

chriswillott, 2021-08-09:
I am running jwst pipeline version 1.3.0 with stcal==0.2.3 on a Mac with os version Catalina 10.15.7. 

When I run ramp fitting without multiprocessing it runs fine.  But when I set ramp_fit.maximum_cores = 'half' the pipeline does not run to completion. It looks like there is a loop that creates multiple processes of the main routine so that all the pipeline steps get run multiple times and the machine memory gets consumed rapidly. I have googled and found some issues around how multiprocessing processes are started (using either spawn or fork) in more recent versions of macOS. 

I do not see this problem when running the same code on a linux machine.

Chris Willott

jdavies-st, 2021-08-09:
So `jwst 1.3.0` does not use `stcal` at all for `ramp_fitting`.  And fact no release versions of `jwst` do, yet.  See:

https://github.com/spacetelescope/jwst/blob/4235b1dd5ea7f9bf457af68c564bcb8ad6ecd7ce/jwst/ramp_fitting/ramp_fit_step.py#L5

If you are using a dev version of `jwst` install the latest, as we are in the middle of making changes, and we have done some internal API changes in ramp fit multiprocessing.  For a long while, multiprocessing in ramp fitting was broken on our master branch.  It should working today.

If not, let us know!  Traceback would be useful too.

chriswillott, 2021-08-09:
moving to jwst

closed 2021-08-09 by chriswillott