spacetelescope/stcal #103: ramp_fitting using the 'saturated' flag to suppress one group ramp fits

cshanahan1, 2022-06-21:
I found a potential issue with using the 'saturation' flag to mark pixels with just one good group to suppress ramp fitting.

If you have a pixel that has DQ flags:
      [0 4 4 4 4 4 4 2 2 2]
 these will become
       [2 4 4 4 4 4 4 2 2 2]
since there is only one usable group left. This triggers the `do_all_sat` function to be called since the 0th group is saturated, 
(On line 835 of `ols_fit.py':
`  first_gdq = groupdq[:, 0, :, :] `
    `if np.all(np.bitwise_and(first_gdq, ramp_data.flags_saturated)):`
        `image_info, integ_info, opt_info = utils.do_all_sat(`
            `ramp_data, inpixeldq, groupdq, imshape, n_int, save_opt)`_

Assuming that the 0th group means that all subsequent groups are saturated is only true when the flag is applied for actual saturation (since that is what the saturation routine does). This isn't true when the flag is being used as a placeholder to suppress ramp fitting. If, for example, your DQ flags were:
       [4 0 4 4 4 4 4 2 2 2]
Then the function to flag solitary good groups would set it to:
       [4 2 4 4 4 4 4 2 2 2]
The `do_all_sat` function wouldn't be triggered since the 0th group isn't saturated, even though it's the same scenario as above. 

I think the `do_all_sat` function should really be a `do_all_no_good_groups` function, and the check should be against saturation, jump, and donotuse flags rather than just assuming if the 0th is saturated you have no good groups left. 

I have a branch and can open a PR for this @kmacdonald-stsci unless theres a reason this is a non-issue?

cshanahan1, 2022-06-21:
@kmacdonald-stsci

kmacdonald-stsci, 2022-06-21:
@cshanahan1 The PR https://github.com/spacetelescope/stcal/pull/99 changes the way one good group ramps get suppressed.  How these ramps should be suppressed is still being discussed.