spacetelescope/stcal #113: M1 multiprocessing issues

kcrasmus, 2022-07-29:
Hello,

     I'm using the Eureka pipeline (with a Mac OSX M1 chip), which calls on stcal tools; in particular, the ols_fit.py file. My code is stalling without error messages when it gets to this line: 

number_slices = utils.compute_slices(max_cores)

I got around this by setting number_slices to 8, but then it stalls again at 

pool = Pool(processes=number_slices)
pool_results = pool.starmap(ols_ramp_fit_single, slices)

I have tried fixing this with
 
from multiprocessing import get_context
pool = get_context("fork").Pool(processes=8)
pool_results = pool.starmap(ols_ramp_fit_single,slices)

but that also stalls. Hopefully someone here knows more than me about working around M1 chip issues :~)

nden, 2023-03-15:
@kcrasmus Sorry this fell through the cracks. There were issues on M1 which I believe were resolved though I'm not sure about this one. Would you be able to confirm if this is still a problem?

braingram, 2024-04-23:
Thanks for opening the issue.

I have a hunch this is related to:
https://github.com/spacetelescope/stcal/issues/247

The above linked issue was addressed (and closed). I will close this issue but please feel free to re-open it if you're still encountering problems.

closed 2024-04-23 by braingram