spacetelescope/stcal #120: Bug in comparison of science and dark ramp lengths in stcal/dark_current/dark_sub.py

chriswillott, 2022-10-03:
Line 99 of https://github.com/spacetelescope/stcal/blob/main/src/stcal/dark_current/dark_sub.py is:

`sci_total_frames = sci_ngroups * (sci_nframes + sci_groupgap)`

but it should be

`sci_total_frames = sci_ngroups * sci_nframes + (sci_ngroups - 1) * sci_groupgap`

because there is one less gap than the number of groups.

This was discovered using NIRCam DEEP8 readout with sci_ngroups = 7. The dark reference file has 130 groups which is 2 frames more than the science data. However, the pipeline refused to attempt dark subtraction because the error above led to:
`Not enough data in dark reference file to match to science data.`

hbushouse, 2022-10-04:
This problem is now being tracked in https://jira.stsci.edu/browse/JP-2938

hbushouse, 2022-10-04:
@chriswillott I assume the same updated math should also apply to the next line of code, where the same info is calculated for the dark data in the reference file, although it's probably meaningless given that the convention is to always have the dark ref files based on data that use nframes=1 and groupgap=0. But I updated that line too, just for consistency. See #121

chriswillott, 2022-10-04:
Yes.

PR #121 (2022-10-04): JP-2938: Fix total frames computation in dark current

Resolves [JP-2938](https://jira.stsci.edu/browse/JP-2938)

Closes #120 

This PR fixes a bug in the computation of the total number of data frames in science and dark ramps when nframes > 1 and/or groupgap > 0.

**Checklist**
- [x] added entry in `CHANGES.rst` (either in `Bug Fixes` or `Changes to API`)
- [ ] updated relevant tests
- [ ] updated relevant documentation
- [ ] updated relevant milestone(s)
- [x] added relevant label(s)
closed 2022-10-04 by hbushouse