spacetelescope/stcal #152: max on an empty array raises a valueerror

zerothi, 2023-03-14:
In some cases when this line encounters a boolean array that has no true values the resulting array will have a 0 length, and then `max` will raise a valueerror.

I have a user which encounters this problem, but I am not fully sure why it happens.

https://github.com/spacetelescope/stcal/blob/6eff679acd6cf382b54fd0559b530d660dc5cc94/src/stcal/ramp_fitting/ols_fit.py#L3454

nden, 2023-03-15:
@zerothi Thanks for the report. Someone will look into it and post back here.

zerothi, 2023-03-16:
I can confirm that this happens when the multiprocessing pool exceeds the dimension it crosses. For instance our example ran on 20 cores but with an image of 16x512 pixels, this resulted in a problem for 4 cores which had 0 elements to calculate. Probably this should be handled upstream (in jwst).

kmacdonald-stsci, 2023-03-16:
> I can confirm that this happens when the multiprocessing pool exceeds the dimension it crosses. For instance our example ran on 20 cores but with an image of 16x512 pixels, this resulted in a problem for 4 cores which had 0 elements to calculate. Probably this should be handled upstream (in jwst).

This arises in STCAL only and not upstream.  When multiprocessing is invoked, the data is sliced up by rows, so running 20 cores on data with 16 rows leaves 4 cores operating on nothing.  The change will be in STCAL to ensure the actual number of processors used is no greater than the number of rows in the dataset, regardless of the number of processors requested.

zerothi, 2023-03-16:
Ok!

PR #154 (2023-03-17): JP-3132: Errors During Multiprocessing When More Processors are Requested than Rows in the Data

Resolves [JP-3132](https://jira.stsci.edu/browse/JP-3132)

Closes #152 

This PR addresses errors during processing when invoking multiprocessing requesting more processors than rows.  The data is sliced up by dividing the number of rows by the number of processors requested.  If the number of rows is less than the number of processors requested, this results in processors running on empty data sets with dimensions (nints, ngroups, 0, ncols).

To prevent this from occurring safeguards have been imp [... cut]
closed 2023-03-17 by kmacdonald-stsci