spacetelescope/stcal #170: Optimization: roughly 40% of a call to `ols_ramp_fit_single` is consumed by 2 lines

braingram, 2023-05-22:
https://github.com/spacetelescope/stcal/blob/main/src/stcal/ramp_fitting/ols_fit.py#L1657-L1658

These appear to be iterating through pixels. Is there a way to convert this to a vector operation or some other optimization?

Here is a plot of the time taken to run `ols_ramp_fit_single` generated with [snakeviz](https://jiffyclub.github.io/snakeviz/) (note the 40% number was calculated using [line_profiler](https://github.com/pyutils/line_profiler) as snakeviz times don't always make sense, note that `calc_slope` contains a block for `calc_slope` with an equal measurement of time).
[image]

zacharyburnett, 2024-07-01:
the link in the discussion is not pinned to a specific commit, so the line numbers have changed since original posting: https://github.com/spacetelescope/stcal/blob/3fa4d42183fb384166d51f1aca8dd7bbe0ca6603/src/stcal/ramp_fitting/ols_fit.py#L1899-L1900

braingram, 2024-07-30:
This can likely be closed since there is now a c version (and the python version will likely be removed).

I'm leaving it open for now but if anyone feels like it should be closed feel free to close it.

closed 2025-02-14 by braingram