spacetelescope/stcal #347: Possible duplicate reproject() function

emolter, 2025-02-25:
Is there any meaningful difference between `stcal.alignment.util.reproject()` and `stcal.outlier_detection.utils.reproject()`?  Could the one in outlier detection be removed?  @mcara @nden @braingram

mcara, 2025-02-25:
is it used anywhere?

mcara, 2025-02-25:
`calc_gwcs_pixmap` and `calc_pixmap` are no longer used. these were the main clients. All those utils modules need a clean-up based on what is used.

emolter, 2025-02-25:
`calc_gwcs_pixmap` is used inside `gwcs_blot`, which is called from the outlier detection step in both `jwst` and `romancal`.  That function currently uses the `reproject` version that is in `stcal.outlier_detection.utils`.

I agree RE `calc_pixmap` and the `reproject` version that is in `stcal.alignment.util` - these appear unused.

There is also a `calc_gwcs_pixmap` and `reproject` in Romancal: https://github.com/spacetelescope/romancal/blob/7484c3870708e5b00ab389e5ade94150373fd9fe/romancal/resample/resample_utils.py#L253, and that version of `calc_gwcs_pixmap` is currently used in two places in Romancal's resample step.  It appears that those will be removed by https://github.com/spacetelescope/romancal/pull/1634, and Romancal will start calling `gwcs_blot` from `stcal.outlier_detection.utils`.

None of the three `reproject` functions (Romancal, alignment, outlier_detection) are identical, although it looks to me like they accomplish the same thing.

I would suggest that we decide which of these is the "best" and then remove the others.  Do you have any opinion on which implementation is preferred?